espressif/esp-usb #19: [UVC] Get hub hardware port with device before UVC_OPEN

lijunru-hub, 2024-02-28:
On the IDF branch hotfix/usb_hub_uninstall, a new field for the hardware port of the hub has been added. However, currently, this information can only be retrieved within UVC_OPEN. It is desired to add an API to retrieve device information before UVC_OPEN.

eg:
`esp_err_t libuvc_get_device_info(uvc_device_t *dev, usb_device_info_t *dev_info)
{
    uvc_error_t ret;
    struct libusb_device_handle *usb_devh;

    ret = libusb_open(dev->usb_dev, &usb_devh);
    UVC_DEBUG("libusb_open() = %d", ret);
    if (ret != UVC_SUCCESS) {
        return ESP_FAIL;
    }

    uvc_camera_t *camera = (uvc_camera_t *)(usb_devh);
    RETURN_ON_ERROR( usb_host_device_info(camera->handle, dev_info) );

    libusb_close(usb_devh);
    return ESP_OK;
} `

tore-espressif, 2024-03-07:
@lijunru-hub Sorry for the late reply.

Draft implementation is here https://github.com/espressif/esp-usb/pull/24/commits/a639a34e904cc0a3ba6d10ab9c7d3b6341c4cbc6

Compared to your proposal, it is a little different: It accepts a `uvc_device_handle_t` , so the UVC device must be already opened. Does this solve your problem?

lijunru-hub, 2024-03-08:
I think before uvc_open to get the hardware's info is better.

tore-espressif, 2024-03-08:
> I think before uvc_open to get the hardware's info is better. Like use the uvc_device_t handle

Thank you for the feedback, I understand now. Implemented according to your suggestion in 83cf7019fcb0d49197aedc89460a6c4114e944e5

PR #24 (2024-03-11): UVC driver release 1.0.3
- Added support for ESP32-P4
- Bumped libuvc version to relax frame format negotiation
- Fixed crash on opening non-UVC devices

Related
* libuvc PR https://github.com/libuvc/libuvc/pull/273
* Closes #19
closed 2024-03-11 by tore-espressif