espressif/esp-usb #26: usb hid device barcode scanner can not use

yel-best, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

* usb_host_lib 
* 
After reading the barcode gun, 57 hungry key_code appears directly, and it enters a dead loop

[image]

I'm using this example
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/hid

Please tell me some ways or reasons,thanks

peter-marcisovsky, 2024-03-14:
Hello @yel-best 

Could you please specify and elaborate what do you mean by a following quote:

> it enters a dead loop

Also, could you please specify all the changes, if any, you have made to the HID example code? 
Thank you.

yel-best, 2024-03-15:
> Hello @yel-best
> 
> Could you please specify and elaborate what do you mean by a following quote:
> 
> > it enters a dead loop
> 
> Also, could you please specify all the changes, if any, you have made to the HID example code? Thank you.

Thank you for your reply. Yes, you are right. I submitted this question repeatedly because I was really troubled. I hope you can understand. Thanks again.

Oh, I didn't modify the logic code, I just added some esp_log, and I printed out every function execution point that the program ran. Then I found that the content of my **QR code was `"a" `but the result of scanning the code was keyCode = 57**, and in the function:
```
hid_host_interface_callback
hid_host_keyboard_report_callback
hid_host_interface_callback
hid_host_keyboard_report_callback
key_event_callback 57
```
The loop jump does not output the result character `"a" `I want.

I don’t know the reason, but the result I understand from the code scanner is actually characters such as abcdefg in the keyboard. Do I need to modify anything? I think the driver and character set in hid_usage_keyboard can be applied to the code scanner.

roma-jam, 2024-03-15:
Hi @yel-best ,

 
The regular barcode scanner works a keyboard, so during the scanning it emulates the keyboard key pressing.
That means that example should not do anything, when you are not scanning and show any data, when you scan the barcode. Is that right? 

Then, example parses the data from scanner based on the default keyscan table, which is provided in the example [hid_host_example.c#L90](https://github.com/espressif/esp-idf/blob/4f900edb06b29c1489ea9ae5318ce3ee261b7939/examples/peripherals/usb/host/hid/main/hid_host_example.c#L90). 

In the `line 386, hid_host_example.c` file: 

```
...
    case HID_HOST_INTERFACE_EVENT_INPUT_REPORT:
        ESP_ERROR_CHECK(hid_host_device_get_raw_input_report_data(hid_device_handle,
                                                                  data,
                                                                  64,
                                                                  &data_length));

        ESP_LOG_BUFFER_HEX(TAG, data, data_length); // This is debug output for raw data input report

        if (HID_SUBCLASS_BOOT_INTERFACE == dev_params.sub_class) {
...
```
Thanks.

yel-best, 2024-03-15:
> hid_host_device_get_raw_input_report_data

yes, you are right. The code scanner is actually simulating the keyboard for data output. I cannot recognize the USB HID device after running the example. I suspect that the code scanner I use is not standard enough. I will try to replace it with another one first. Scan code gun, and then add this code, I will update the log, please wait until I find a new code scanner, I will update again

thank you for your help

JeevabharathiRK, 2024-03-17:
Hi @yel-best,
<ul>
<li>I like to share my experience in making a connection with the **Barcode scanner and ESP32 S3 Devkit M1** board, I used to upload code via Visual Studio Code, but I had an issue there, so I searched for that and found that it could be solved by using Official Espressif IDE.</li>
<li>I am using Espressif-IDE Version: 2.12.0 and example code-named "hid_host_example.c"<br>
[image][image]</li>
<li>Connection was as per the readme file and additionally, I have shared some images for better understanding. The connected barcode scanner was brought from Amazon.
<br>
[image]
[image]<br>
[image]
[image]</li>
<li>And Here is the Output ..... <br>
[image]
[image]

</li>

yel-best, 2024-03-18:
> Hi @yel-best,
> 
> * I like to share my experience in making a connection with the **Barcode scanner and ESP32 S3 Devkit M1** board, I used to upload code via Visual Studio Code, but I had an issue there, so I searched for that and found that it could be solved by using Official Espressif IDE.
> * I am using Espressif-IDE Version: 2.12.0 and example code-named "hid_host_example.c"
>   <img alt="" width="300" src="https://private-user-images.githubusercontent.com/67570918/313482098-ea3d8415-815e-4e38-88bf-6ba3883ba32a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3Mjk3MzAsIm5iZiI6MTcxMDcyOTQzMCwicGF0aCI6Ii82NzU3MDkxOC8zMTM0ODIwOTgtZWEzZDg0MTUtODE1ZS00ZTM4LTg4YmYtNmJhMzg4M2JhMzJhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyMzcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEzNDJlY2NlMTMyZGEyMGVkNWFmOTI5ZjNkZmQwZTFlNzBmNmJhM2M0NGFmMjgzODA1ZmIxMTliOGI4YWY0YjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.J6NV7DHXVtJfTFqBO8P2Qc57znZEKMPmwDQoTg0WInM"><img alt="" width="200" src="https://private-user-images.githubusercontent.com/67570918/313482271-65b6f32e-ca0c-4a11-b8d0-0f7e5dc289cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3Mjk3MzAsIm5iZiI6MTcxMDcyOTQzMCwicGF0aCI6Ii82NzU3MDkxOC8zMTM0ODIyNzEtNjViNmYzMmUtY2EwYy00YTExLWI4ZDAtMGY3ZTVkYzI4OWNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyMzcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5ZGE1NjllNDkxMzYzZWQ5YzA5NmFiNWUwMzAzYmM3MWExYjMxYTc5ZjIxNjVlNTUxODdhYWI4YzI1NDA5NDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.y6Bwq0bYMbnsfccsjSvzzb3ROpn-FTPpsZyrSUZSL9w">
> * Connection was as per the readme file and additionally, I have shared some images for better understanding. The connected barcode scanner was brought from Amazon and It was dismantled.
>   
>   <img alt="" width="200" src="https://private-user-images.githubusercontent.com/67570918/313485904-20d530bb-b301-4dbd-b9e9-edc60511a59a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3Mjk3MzAsIm5iZiI6MTcxMDcyOTQzMCwicGF0aCI6Ii82NzU3MDkxOC8zMTM0ODU5MDQtMjBkNTMwYmItYjMwMS00ZGJkLWI5ZTktZWRjNjA1MTFhNTlhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyMzcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ5NjY0YjA5ODIxNTVjN2IwYWE0Y2NjNDg3MWRmM2EwNzA4NjZmYjFlYzJmYjBiNGU3YzcxNjYyOThlZDQ3MTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.S3FTizVGS_RXrndPJTu6eUzgGudq2F7D98wUomlMSGs">
>   <img alt="" width="200" src="https://private-user-images.githubusercontent.com/67570918/313486031-e55ea337-bd35-408c-bafc-053f5b4a88b2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3Mjk3MzAsIm5iZiI6MTcxMDcyOTQzMCwicGF0aCI6Ii82NzU3MDkxOC8zMTM0ODYwMzEtZTU1ZWEzMzctYmQzNS00MDhjLWJhZmMtMDUzZjViNGE4OGIyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyMzcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI1YzBkYjJiODlkYTkzYjEzMzM5Njc4MTcyNTlkNTA3MDRiYWRkMDRkMjJiZDk0YzVmMzdkYzg1ZmM2NWViNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.oCvGqopEuIqj0MUaNFP06SVEKOlnDDPmZZRFFdAKfqc">
>   <img alt="" width="200" src="https://private-user-images.githubusercontent.com/67570918/313486345-b663f20a-1d70-475b-9adf-ac61a2fbd1b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3Mjk3MzAsIm5iZiI6MTcxMDcyOTQzMCwicGF0aCI6Ii82NzU3MDkxOC8zMTM0ODYzNDUtYjY2M2YyMGEtMWQ3MC00NzViLTlhZGYtYWM2MWEyZmJkMWI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyMzcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4MDNhNTg0OTk1NWE4NTc4NjY3MGQzNWMyMjJjZGViYjhkZjhiZDM1ZmJjZGU0Njk5ZDFlZDAyOTc1NTM4ZDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4K3nl_Gz1pHkeFYkpNTaigspyZjGnyZaFdMScKOQPHY">
>   <img alt="" width="200" src="https://private-user-images.githubusercontent.com/67570918/313486732-cee20f29-0eb4-4c43-98b1-ae8c33937f4f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3Mjk3MzAsIm5iZiI6MTcxMDcyOTQzMCwicGF0aCI6Ii82NzU3MDkxOC8zMTM0ODY3MzItY2VlMjBmMjktMGViNC00YzQzLTk4YjEtYWU4YzMzOTM3ZjRmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyMzcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1MTkzZDc [... cut ...] i1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyMzcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTllNjU0NzE1Y2Y5OWYzMDRlNjE2MGU1ZWIyMmU3NGI4ZTBhOTU0ODEzNTc4OTE4OGY4ZTRmZGNmZTM0NTdjNzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vVdqwbzgE0gdPSBRybQOiwg7EiYV64OAnRNtHJ1ZD8Y">

oh, thank you very much for sharing, you are so kind, this will bring me a solution, thank you very much 😄

roma-jam, 2024-03-19:
Hey @yel-best,
Are there any updates? 

> I cannot recognize the USB HID device after running the example. I suspect that the code scanner I use is not standard enough. 

Based on the debug output from hid example you had provided, it is a USB device that has one Keyboard interface. So the example found the "keyboard" and handle the key scan code. So, it should work. 

If there anything new, please feel free to report.

roma-jam, 2024-04-01:
Hi @yel-best, 
Are there any progress?

yel-best, 2024-04-02:
> Hi @yel-best, Are there any progress?

Thank you for your concern. After trying it, I suspected that the protocol of the scanner hardware may not be standard. I suspected that it was a hardware compatibility issue. Then we replaced two scanner devices for USB connection. It worked without any code changes. Correctly obtain the corresponding barcode or QR code value, so this is how I currently solve my problem.

thanks 😄

roma-jam, 2024-04-02:
@yel-best 
Thanks for the information.

I will close the issue then. If there will be anything else, feel free to re-open that issue or create a new one.

closed 2024-04-02 by roma-jam