espressif/esp-usb #28: CDC class stopped working on VirtualBox

higaski, 2024-04-12:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

v5.2

### Development Kit.

custom board

### Used Component version.

1.4.4

### More Information.

I feel like I need to apologize in advance, because this is one weird issue and I can hardly give any information.

I'm using esp_tinyusb from the Espressif registry, currently in version 1.4.4. I have an application which creates a CDC interface and connects to some legacy application written in Visual Basic 6.0. Since I'm running Linux I have to run this application inside a Windows 10 VM in VirtualBox (7.0.14). This used to work.

For some reason though it stopped working (can't say when exactly) and now I can't even connect the USB device to my Windows 10 VM without getting an error from tinyusb:
```sh
No endpoints available (ep_max=6) 

cdcd_open 305: ASSERT FAILED

process_set_config 967: ASSERT FAILED

process_control_request 717: ASSERT FAILED

No endpoints available (ep_max=6) 

cdcd_open 305: ASSERT FAILED

process_set_config 967: ASSERT FAILED

process_control_request 717: ASSERT FAILED
```

I've tried various other virtual machines so I assume the issue has something to do with VirtualBox. Other devices I use where I know that they create CDC classes work though.

On my host machine the CDC class also seems to work just fine. I can send and receive UART strings without any issues whatsoever. I'd greatly appreciate clues on where to even start looking... at the moment I have absolutely no idea.

roma-jam, 2024-04-14:
Hi @higaski , 

May I ask you to revert to the previous tinyusb version? 

You can do this by requiring specific version in your main/idf_component.yml like this:

```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp_tinyusb: "^1.4.2"
  espressif/tinyusb: '0.15.0~5'
```

We will fix that issue ASAP.

higaski, 2024-04-14:
Thanks for figuring that out so fast, I'll try that first thing tomorrow morning.

roma-jam, 2024-04-18:
Hi @higaski , 

Thanks for reporting the issue and very detailed description!

We have released the fix, feel free to try out new version (which is 0.15.0~8).

Thanks

higaski, 2024-04-18:
Thanks for the fast fix. Will there be an esp_tinyusb release as well?

roma-jam, 2024-04-18:
@higaski  

No, there is nothing to do with esp_tinyusb, so the version is the same.

Let me know, is there will be any problems with last tinyusb.
Otherwise, feel free to close the issue.

higaski, 2024-04-18:
Looks good. One last question. What's your opinion (or maybe Espressifs internal best practice?) on how to handle nested component dependencies in such a case as tinyusb?

Would you recommend to explicitly specify both versions like so
```yml
dependencies:
  espressif/esp_tinyusb: "^1.4.2"
  espressif/tinyusb: '0.15.0~5'
```

or rather just the esp_tinyusb one and let it "figure out the rest"?
```yml
dependencies:
  espressif/esp_tinyusb: "^1.4.2"
```

roma-jam, 2024-04-18:
@higaski , 

IDF component manager provides the possibility of handling the component dependencies pretty flexible (more information [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-component-manager.html))

Right now, there is no need to specify both versions in the manifest (if you don't need to use the fixed versions always) . More about versioning [here](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/versioning.html#versioning-scheme)).

I will close the issue. If there is something else I could help, please let me know or is there will be anything else - feel free to re-open this issue or create a new one. 

Thanks.

closed 2024-04-18 by roma-jam