espressif/esp-usb #38: MSC: FATFS isn't remounted after USB disconnect

toine512, 2024-06-09:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

5.2.1

### Development Kit.

ESP32-S3-DevKitC-1 clone

### Used Component version.

1.4.4

### More Information.

Hello.

Using the flash to MSC flow, the FATFS partition is not remounted after USB host is disconnected because of this abandoned TinyUSB bug: https://github.com/hathach/tinyusb/issues/1592 .
The symptom is that my mount callback passed to `storage_init_spiflash()` is never called besides startup when I call `tinyusb_msc_storage_mount()`, while no mount error message appears either.
I verified this is the issue by adding the callback in tusb_msc_storage.c as a copy/paste of tud_umount_cb():
```C
// Invoked when device is unmounted
void tud_umount_cb(void)
{
    if (tinyusb_msc_storage_mount(s_storage_handle->base_path) != ESP_OK) {
        ESP_LOGW(TAG, "tud_umount_cb() mount Fails");
    }
}

void tud_suspend_cb(bool none)
{
    if (tinyusb_msc_storage_mount(s_storage_handle->base_path) != ESP_OK) {
        ESP_LOGW(TAG, "tud_umount_cb() mount Fails");
    }
}
```

roma-jam, 2024-06-09:
Hi @toine512, 

May I ask you to: 
1. clarify the version of tinyusb component version you are using (should be something like 0.15.x.x)
2. have you tired to launch original `tusb_msc` example from `esp-idf` ? 
3. for device to be able to detect connect/disconnect events from usb, the several additional steps should be fulfilled (hardware resistor divider on Vbus + software configuration during `tusb_install()` call. More information here:  [Self-Powered Device](https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32s2/api-reference/peripherals/usb_device.html#self-powered-device)

toine512, 2024-06-09:
1. I'm using esp_tinyusb 1.4.4 package. By digging into automatically downloaded components I find tinyusb version is 0.15.1. I use PlatformIO, it ships a crippled ESP-IDF, managing components is partly manual.
2. No. I didn't write any tinyusb code anyway: everything is handled by the call to `tinyusb_msc_storage_init_spiflash()`.
3. You're absolutely right and I didn't even think of checking that VBUS detection is wired properly on the board. That obviously would explain why it thinks it has been suspended by a SE 0.
I will do necessary checks and get back to you, thanks for your help.

toine512, 2024-06-22:
Hardware and configuration overlook confirmed.

closed 2024-06-22 by toine512