espressif/esp-usb #43: An exception occurs when I use the scanner and call the api after scanning

yel-best, 2024-06-20:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

host/hid

### ESP-IDF version.

v5.1.4

### Development Kit.

ESP32S3 N1RR8

### Used Component version.

1.0.2

### More Information.

use usb_host_hid v1.0.2

I connected a barcode scanner device to the **USB HID** and I added a function in the following method to record the code obtained by the barcode scanner and call an HTTP API for background verification. When I encountered a problem in this process, the following is part of the code

```
static inline void hid_keyboard_print_char(unsigned int key_char)
{
    if (!!key_char)
    {
        // printf("\n %i", qr_code_index);
        if (key_char != 0 && key_char != '\n')
        {
            qr_code[qr_code_index] = (char)key_char;
        }
        putchar(key_char);
        qr_code_index++;
#if (KEYBOARD_ENTER_LF_EXTEND)
        if (KEYBOARD_ENTER_MAIN_CHAR == key_char)
        {
            // putchar('\n');
            printf("\n");
            // printf("QR codes::%s\n", qr_code);
            // qr_code[qr_code_index] = '\0';
            qr_code[strlen(qr_code) - 1] = 0;
            if (lv_set_order(qr_code) == ESP_OK)
            {
                memset(qr_code, 0, sizeof(qr_code));
                qr_code_index = 0;
            }
        }
#endif // KEYBOARD_ENTER_LF_EXTEND
        fflush(stdout);
    }
}
```

`lv_set_order(qr_code)`  is call api logic

```
Generic
GEG24050600751
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (USB HID Host)
Core  0 register dump:
PC      : 0x4038bcb4  PS      : 0x00060336  A0      : 0x4037b291  A1      : 0x3fcb4d30  
0x4038bcb4: _frxt_int_enter at E:/MyProject/IoT/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:125

0x4037b291: _xt_lowint1 at E:/MyProject/IoT/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1233

A2      : 0x00060323  A3      : 0x00000000  A4      : 0x00060320  A5      : 0x3fcb9bc8
A6      : 0x3fcb4e90  A7      : 0x00000000  A8      : 0x80384540  A9      : 0x3fcb4dc0
A10     : 0x3fca5fb4  A11     : 0xffffffff  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00001000  A15     : 0x3fcac1b4  SAR     : 0x0000000e  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
0x40056f5c: memcpy in ROM

0x40056f72: memcpy in ROM

Backtrace: 0x4038bcb1:0x3fcb4d30 0x4037b28e:0x3fcb4e00 0x40389837:0x3fcb4e20 0x4207ba22:0x3fcb4e60 0x4206a246:0x3fcb4e90 0x4207c3f9:0x3fcb4ec0 0x420cbf08:0x3fcb4f00 0x42085881:0x3fcb4f50 0x42085dea:0x3fcb4fe0 0x420860ba:0x3fcb5020 0x42087a58:0x3fcb5060 0x420fb961:0x3fcb5090 0x4208953e:0x3fcb50b0 0x4208a298:0x3fcb50e0 0x4200da12:0x3fcb5110 0x4200dc8f:0x3fcb51e0 0x4200b1fe:0x3fcb52d0 0x42011809:0x3fcb52f0 0x42011a4d:0x3fcb5310 0x4203dfb4:0x3fcb5390 0x420503a6:0x3fcb53b0 0x4203deed:0x3fcb53e0
0x4038bcb1: _frxt_setup_switch at E:/MyProject/IoT/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:104

0x4037b28e: _xt_lowint1 at E:/MyProject/IoT/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1225

0x40389837: xQueueGenericSend at E:/MyProject/IoT/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:998

0x4207ba22: sys_mbox_post at E:/MyProject/IoT/esp/esp-idf/components/lwip/port/freertos/sys_arch.c:239

0x4206a246: tcpip_send_msg_wait_sem at E:/MyProject/IoT/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 2)

0x4207c3f9: netconn_gethostbyname_addrtype at E:/MyProject/IoT/esp/esp-idf/components/lwip/lwip/src/api/api_lib.c:1325 (discriminator 2)

0x420cbf08: lwip_getaddrinfo at E:/MyProject/IoT/esp/esp-idf/components/lwip/lwip/src/api/netdb.c:345

0x42085881: getaddrinfo at E:/MyProject/IoT/esp/esp-idf/components/lwip/include/lwip/netdb.h:23
 (inlined by) esp_tls_hostname_to_fd at E:/MyProject/IoT/esp/esp-idf/components/esp-tls/esp_tls.c:199
 (inlined by) tcp_connect at E:/MyProject/IoT/esp/esp-idf/components/esp-tls/esp_tls.c:336

0x42085dea: esp_tls_low_level_conn at E:/MyProject/IoT/esp/esp-idf/components/esp-tls/esp_tls.c:436

0x420860ba: esp_tls_conn_new_sync at E:/MyProject/IoT/esp/esp-idf/components/esp-tls/esp_tls.c:522

0x42087a58: ssl_connect at E:/MyProject/IoT/esp/esp-idf/components/tcp_transport/transport_ssl.c:111

0x420fb961: esp_transport_connect at E:/MyProject/IoT/esp/esp-idf/components/tcp_transport/transport.c:123

0x4208953e: esp_http_client_connect at E:/MyProject/IoT/esp/esp-idf/components/esp_http_client/esp_http_client.c:1396

0x4208a298: esp_http_client_open at E:/MyProject/IoT/esp/esp-idf/components/esp_http_client/esp_http_client.c:1553

0x4200da12: app_terminal_api_common_auth at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_lvgl/app_lvgl_http.c:469

0x4200dc8f: terminal_http_api_request at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_lvgl/app_lvgl_http.c:559

0x4200b1fe: lv_set_order at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_lvgl/app_terminal_ui.c:828

0x42011809: hid_keyboard_print_char at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_usb/app_usb_scanner.c:255
 (inlined by) key_event_callback at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_usb/app_usb_scanner.c:281

0x42011a4d: key_event_callback at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_usb/app_usb_scanner.c:276
 (inlined by) hid_host_keyboard_report_callback at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_usb/app_usb_scanner.c:346
 (inlined by) hid_host_interface_callback at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/components/app_usb/app_usb_scanner.c:431

0x4203dfb4: hid_host_user_interface_callback at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/managed_components/espressif__usb_host_hid/hid_host.c:331
 (inlined by) in_xfer_done at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/managed_components/espressif__usb_host_hid/hid_host.c:712

0x420503a6: _handle_pending_ep at E:/MyProject/IoT/esp/esp-idf/components/usb/usb_host.c:621
 (inlined by) usb_host_client_handle_events at E:/MyProject/IoT/esp/esp-idf/components/usb/usb_host.c:767

0x4203deed: hid_host_handle_events at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/managed_components/espressif__usb_host_hid/hid_host.c:1296
 (inlined by) event_handler_task at E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/QA-mesh-terminal/managed_components/espressif__usb_host_hid/hid_host.c:169

```

I don't know why there is such an error, I hope you can provide some help, thanks

tore-espressif, 2024-06-24:
Hello @yel-best , It seems that HID task has stack overflow. Can you increase the stack size?

You can do that from HID install function, using this value https://github.com/espressif/esp-usb/blob/master/host/class/hid/usb_host_hid/include/usb/hid_host.h#L104

yel-best, 2024-06-26:
> Hello @yel-best , It seems that HID task has stack overflow. Can you increase the stack size?
> 
> You can do that from HID install function, using this value https://github.com/espressif/esp-usb/blob/master/host/class/hid/usb_host_hid/include/usb/hid_host.h#L104

Hi tore

Below is the config I created for HID

```
    const hid_host_driver_config_t hid_host_driver_config = {
        .create_background_task = true,
        .task_priority = 4,
        .stack_size = 3072,
        .core_id = 0,
        .callback = hid_host_device_callback,
        .callback_arg = NULL};

    ESP_ERROR_CHECK(hid_host_install(&hid_host_driver_config));

StaticTask_t xHIDInitscanTaskBuffer;
EXT_RAM_BSS_ATTR StackType_t xHIDInitscanStack[3072];

    xTaskCreateStatic(
        usb_lib_task,                // Function that implements the task.
        "usb_events",                // Text name for the task.
        3072,                        // Stack size in bytes, not words.
        xTaskGetCurrentTaskHandle(), // Parameter passed into the task.
        5,
        // tskIDLE_PRIORITY,            // Priority at which the task is created.
        xHIDInitscanStack,           // Array to use as the task's stack.
        &xHIDInitscanTaskBuffer);    // Variable to hold the task's data structure.

StaticTask_t xHIDscanTaskBuffer;
EXT_RAM_BSS_ATTR StackType_t xHIDscanStack[2048];

    xTaskCreateStatic(
        hid_host_task,        // Function that implements the task.
        "hid_host_task",      // Text name for the task.
        2048,                 // Stack size in bytes, not words.
        NULL,                 // Parameter passed into the task.
        5,
        // tskIDLE_PRIORITY,     // Priority at which the task is created.
        xHIDscanStack,        // Array to use as the task's stack.
        &xHIDscanTaskBuffer); // Variable to hold the task's data structure.

```

I don't know whether this occupies **internal RAM or external RAM. The ESP32S3 I use is the N16R8 version.** I want to use external RAM instead of HID RAM. There will be enough space in the external RAM.

`hid_host_install`  Can this hid install be migrated to external RAM?

yel-best, 2024-06-28:
@tore-espressif  Hi  pls help me, thanks  🥲

leeebo, 2024-06-28:
@yel-best  `EXT_RAM_BSS_ATTR StackType_t xHIDscanStack[2048];` using external RAM, please increase the size of the array (try 4096 first)

yel-best, 2024-07-04:
> 4096

Hi @leeebo 

Ok, I will try to modify it to `xHIDscanStack[4096]`  thanks

yel-best, 2024-07-08:
When I call some functions in HID's Event, such as calling the API after scanning the QR code, and then registering the content of the scanned QR code on the server, the following error will appear. I don't understand the reason. I can allocate HID_device to PSRAM. Running?

```
    const hid_host_driver_config_t hid_host_driver_config = {
        .create_background_task = true,
        .task_priority = 4,
        .stack_size = 3072,
        .core_id = 0,
        .callback = hid_host_device_callback,
        .callback_arg = NULL};

    ESP_ERROR_CHECK(hid_host_install(&hid_host_driver_config));
```

```
uru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (USB HID Host)
Core  0 register dump:
PC      : 0x4037b271  PS      : 0x00060136  A0      : 0x3fcb5480  A1      : 0x3fcb53c0  
0x4037b271: _xt_lowint1 at E:/MyProject/IoT/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1205

A2      : 0x00060123  A3      : 0x00000000  A4      : 0x00060120  A5      : 0x00000028
A6      : 0x00000006  A7      : 0x3fcb0fa4  A8      : 0x8038bb9e  A9      : 0x3fcb5490
A10     : 0x00060123  A11     : 0x00000000  A12     : 0x00060120  A13     : 0x3fcb55f0
A14     : 0x000001d0  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
0x400570e8: memset in ROM

0x400570f3: memset in ROM

Backtrace: 0x4037b26e:0x3fcb53c0 0x3fcb547d:0x3fcb5490 |<-CORRUPTED
```

yel-best, 2024-07-08:
This happens when I use a USB HID scanner that automatically sleeps. After disconnecting and reactivating, an error will appear. What is the reason? seems to be an internal error

```
I (369808) app_usb: HID Device, protocol 'KEYBOARD' DISCONNECTED
I (369808) app_usb: HID Device, protocol 'NONE' DISCONNECTED
I (379035) app_usb: hid_host_device_event
I (379035) app_usb: HID Device, protocol 'KEYBOARD' CONNECTED
I (379035) app_usb: hid_host_device_event
I (379040) app_usb: HID Device, protocol 'NONE' CONNECTED
E (379046) hid-host: hid_host_interface_claim_and_prepare_transfer(624): Unable to claim Interface
E (379055) hid-host: hid_host_device_open(1237): Unable to claim interface
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x42010598
```

roma-jam, 2024-07-08:
Hi @yel-best , 

If I understood you correctly, you are trying to run some additional logic inside the HID event. If yes, that this is not possible and you need to use another task for that purpose. Please, refer to the example here: [USB Host HID example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid)
In the example, the `app_event_queue` is using to tie the APP task and HID event. 

1. "After disconnecting and reactivating" - does that mean that you disconnect the scanner from the port physically? Or it disconnects by the internal logic automatically, while nobody is using it?
2. Is that possible to share the output with extended log (you need to enable it via `menuconfig`, setting the `Verbose` level of the debug output) during the problem?  

Thanks.

yel-best, 2024-07-17:
> Hi @yel-best ,
>
> If I understood you correctly, you are trying to run some additional logic inside the HID event. If yes, that this is not possible and you need to use another task for that purpose. [...]

Hi @roma-jam 
Yes, I have run some business logic. I have created a new Task and it can run normally after testing. However, there will still be an exception and restart after the HID USB device is physically removed and reinstalled. The trigger points for restart after HID USB removal are as follows:

* When using a wireless HID USB device, the HID device connection will be disconnected after the device automatically sleeps, and a restart will also occur when the HID device is reconnected after waking up
* When it is a wired HID USB device, the device will be restarted when it is unplugged and reinstalled

yel-best, 2024-07-17:
[app_usb_scanner.zip](https://github.com/user-attachments/files/16258204/app_usb_scanner.zip)

This is the code I used，pls check，thanks

roma-jam, 2024-07-17:
Hi @yel-best ,

>  I have created a new Task and it can run normally after testing

Great, good to hear that it is working. 

>  However, there will still be an exception and restart after the HID USB device is physically removed and reinstalled.

Could you please share the exception crash log? Thanks.

roma-jam, 2024-07-30:
Hi @yel-best, 
Any updates?

roma-jam, 2025-01-20:
Hi @yel-best, 

Is the issue still relevant?

yel-best, 2025-01-20:
> Hi [@yel-best](https://github.com/yel-best),
> 
> Is the issue still relevant?

Hi, no la，can be used normally, thank you for your opinions 😄

closed 2025-01-20 by yel-best