espressif/esp-usb #48: Flash MSC over USB is not mounting back to application after unplugging USB cable

Asanga-Viraj, 2024-07-26:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

host/msc

### ESP-IDF version.

v5.2.2

### Development Kit.

ESP32-S3

### Used Component version.

1.4.4

### More Information.

I tested the example tusb_msc over flash. IDF version is 5.2.2 and Module is ESP32-S3. Tested on Win10. 
[https://github.com/espressif/esp-idf/tree/v5.2.2/examples/peripherals/usb/device/tusb_msc](url)

The flash storage media is exposed to the application at the beginning. When connecting USB to Win PC, the PC recognized as MSC on Windows Explorer which Flash storage partition exposes to USB Host and does not expose to the application. Ejecting the media in Explorer returns the expose of flash storage media to application.

The problem is, the Flash storage partition doesn't expose (mount) back to application after disconnecting USB cable directly from the Host without ejecting it. Calling this function "tinyusb_msc_storage_mount(BASE+PATH)" does the mount to application side but couldn't find a way to detect USB cable disconnection event. 

Also, after ejecting media from Win Explorer, the MSC is not showing the dismount of the media on the Explorer. The icon stay as mounted.

Thanks.

peter-marcisovsky, 2024-07-26:
Hi @Asanga-Viraj 

- could you please specify which development board are you using? Or are you using a custom board?
- If I understood correctly, it's a self-powered USB device in your case (meaning, the device is not powered from the USB cable used for data communication)

Asanga-Viraj, 2024-07-28:
Hi @peter-marcisovsky 

Thanks for commenting. I am using a custom board with power supply (12V in to 3V3) and VBUS signal is not using. USB doesn't provide power to the custom board.

peter-marcisovsky, 2024-07-29:
If it's a custom board and a self powered device, do you have a [Vbus monitor](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/usb_device.html#self-powered-device) implemented on your board?

Asanga-Viraj, 2024-07-29:
No, I don't have that implementation.

peter-marcisovsky, 2024-07-30:
Could you please try implementing it and see if it solves your problem?
According to the documentation I have sent, you will just have to add a resistor divider to your board and update the `tinyusb_config_t` accordingly.

Asanga-Viraj, 2024-08-02:
Thanks for the idea. My design doesn't have spare pin for VBUS monitor because I am using RGB LCD, SPIRAM integrated module and some inputs and outputs. But, I will try my best.

Is there any way to detect USB disconnect by voltage level changing in D- D+ signal. I think in very low level, the driver does it. We have to find a way to get that event outside world if it is possible.

peter-marcisovsky, 2024-08-02:
To detect USB disconnection in your case, without the Vbus monitor, you can use [tud_suspend_cb()](https://github.com/espressif/tinyusb/blob/33d36cb23d6f8415d0e8a09d15a5e6cd8c0a6d05/src/device/usbd.h#L79)

But, please keep in mind, that now you will have 2 callbacks mounting the FS to the app. You should be aware of conflicts situations that might happen. And you should take some steps to create such a logic in you code that will prevent any conflict situation from happening.

Again, the best solution would be to use the Vbus monitor for such a functionality.

Asanga-Viraj, 2024-08-15:
Thanks @peter-marcisovsky. This worked for me.

Another issue came in. ESP crashes when plugging USB to PC during my application is accessing and reading a file from the file system. This happens because USB MSC is mounting back file system in to USB MSC regardless of checking file path opened by firmware side. Is this a issue?

peter-marcisovsky, 2024-08-21:
Yes, ti looks like that is the case. Not having the Vbus monitor, there is no event that detects connection, thus there is no event that would change the FS owner. I have mentioned that this kind of conflicting situations might occur in your workaround.

peter-marcisovsky, 2024-08-27:
Hi @Asanga-Viraj
Feel free to close the issue, if you don't have any other questions, or if your problem has been resolved. 
Thank you.

closed 2024-08-28 by Asanga-Viraj