espressif/esp-usb #51: libusb adapter failed to allocate transfer with length: 16384 (IEC-8)

dmitrij9992905, 2024-07-31:
Hello everyone!

I use ESP32S3 and Lenovo camera. PSRAM 8MB, usage is ON.
I still cannot get an image in any format, only broken files.
Camera has been detected, I can get camera info, but I cannot get image from camera because I see the following errors
```
E (6343) libusb adapter: Failed to allocate transfer with length: 16384
Streaming...
Enabling auto exposure ...
E (6343) USBH: Dev 1 EP 0 STALL
E (6353) Transfer failed: Status 4
```

tore-espressif, 2024-07-31:
I'm really sorry that this issues is unanswered for so long. 
ATM, we are working on our custom UVC driver, because there were too many limitations in the original driver based on libuvc.

In this case, the issue is that the USB transfer must be allocated in DMA capable memory, that is, the internal RAM. This limitation will be resolved in the new UVC driver

nicklasb, 2024-12-22:
@tore-espressif When will this be addressed? Is there a workaround that you would recommend?

tore-espressif, 2025-01-31:
Hello everyone, 
We release UVC driver v2 that does not use libuvc anymore.
This gives you extra flexibility to allocate your USB transfers and frame buffers.
Check the driver and the examples here https://github.com/espressif/esp-usb/tree/master/host/class/uvc/usb_host_uvc 
We tested with couple of cameras, so please let us know if you experience any problems.

Indastri, 2025-03-13:
Since there was no response from the user side, closing this issue. If you will encounter further issues please reopen this or create a new issue.

Cheers!

closed 2025-03-13 by Indastri