espressif/esp-usb #54: Speed up tinyusb CDC serial transfer rate on ESP32-S3 (IEC-93)

Superberti, 2024-07-31:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm using an ESP32-S3 on ESP-IDF on the release/5.2 branch.
I made some tests with the tinyusb CDC serial profile but I was not able to get read transfer speeds on the native port (pins 19/20) over 50 KiB/s. The theoretical maximum rate should be something about 1 MiB/s in this mode.
Regardless what I'm doing (larger buffers and FIFOs, using tinyusb with or without tusb_config-h), it had no influence on the transfer speed. So it seems that the speed is limited in some other way. The USB port is connected to a C# test program which sends out the data as fast as it can, so this should not be a problem at all.
Any ideas how to speedup the CDC read speed?

Thanks and bye,
Oliver

tayvs, 2024-11-01:
@tore-espressif Hey.
Are there any updates? Is there an open issue in tinyUSB that provides latest updates?

tore-espressif, 2024-11-01:
The ticket in upstream TinyUSB is here https://github.com/hathach/tinyusb/pull/2576 , it was merged a month ago.

It is already tested on S2 and S3, right now we are preparing P4 support and then we can release new version

emaayan, 2024-12-04:
hi ,will this effect the NCM performance  throughput as well?

roma-jam, 2025-01-20:
Hi @emaayan,

At the current moment the version of `espressif/esp_tinyusb` which enables the DMA mode by default [esp_tinyusb v1.5.0 PR](https://github.com/espressif/esp-usb/pull/96) is available. 

Also, there is an possibility to select DMA mode or to use the previous way of USB data transfer (Slave/IRQ). This also might help you to verify the difference and profile your C# test program.

jrahlf, 2025-07-22:
I want to add, to achieve high throughput (at low cpu load) with the new DMA driver, you should also increase `CONFIG_TINYUSB_CDC_EP_BUFSIZE`. See also my request here: [#212](https://github.com/espressif/esp-usb/issues/212)

closed 2025-09-15 by tore-espressif