espressif/esp-usb #55: USB-HOST-HID connect scanner STALL

Sentaku1129, 2024-08-02:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

host/hid

### ESP-IDF version.

v5.3.0

### Development Kit.

ESP32-S3-WROOM-1

### Used Component version.

1.0.2

### More Information.

I used the development board to connect the scanner gun and the STALL sign appeared. Use "https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/hid" template modification, use the version number is component 1.0.2, I will attach the data collected by connecting the scanner to the S3 and connecting the computer to the USB logic analyzer
[logic.zip](https://github.com/user-attachments/files/16465940/logic.zip)

roma-jam, 2024-08-02:
Hi @Sentaku1129 , 

Could you please update the *.csv log file format or let me know which software could be used to open the *.usmn format. 

Thanks.

Sentaku1129, 2024-08-02:
You can download the program from the URL listed below to open and view the ".usmn" file. The URL is as follows: "https://www.wch.cn/downloads/USB20Monitor_ZIP.html"

Sentaku1129, 2024-08-08:
Some devices will respond to data requests that interrupt IN only if they have sent OUT information about the system's numeric keypad area and case lock status

roma-jam, 2024-08-08:
Hi @Sentaku1129 , 

Sorry for such a delay, I have checked the log file and I was that the STALL'ed transfer is IN transfer with data length = 0. 

Current implementation of HID Host Class Driver is working only with HID devices which are able to work in the BOOT mode (This is specific simplified protocol for devices to be able to work in BIOS e.t.c.).
Any device, which is a bit complicated (such as Gamepads or some scanners) requires more complicated protocol, therefore more complicated structure of input reports.

The example forces the device to change the protocol to BOOT after opening and call `SetIdle(0)` class specific request to disable notification from the device, even without any events (refer to the code [here](https://github.com/espressif/esp-idf/blob/8e4454b285ad39881c5bf3f440d8be617a2f18a8/examples/peripherals/usb/host/hid/main/hid_host_example.c#L439)). 
After these lines, the device will send the BOOT report via IN transfer only when some event has happened, (like key pressing).

You could try to disable this configuration and simplify the logic by removing these lines (`hid_host_example.c`, function `hid_host_device_event()`): 
```
        ESP_ERROR_CHECK(hid_host_device_open(hid_device_handle, &dev_config));
        // Comment out start
        // if (HID_SUBCLASS_BOOT_INTERFACE == dev_params.sub_class) {
        //     ESP_ERROR_CHECK(hid_class_request_set_protocol(hid_device_handle, HID_REPORT_PROTOCOL_BOOT));
        //     if (HID_PROTOCOL_KEYBOARD == dev_params.proto) {
        //         ESP_ERROR_CHECK(hid_class_request_set_idle(hid_device_handle, 0, 0));
        //     }
        // }
        // Comment out stop
        ESP_ERROR_CHECK(hid_host_device_start(hid_device_handle));
``` 

But in this case, the input reports will be not compatible with current parsing logic, so you need to disable the parsing as well.
To verify the Scanner again, you can add the debug for the incoming report (`hid_host_example.c`, function `hid_host_interface_callback()`, event `HID_HOST_INTERFACE_EVENT_INPUT_REPORT`).

So, something like this: 
```
    case HID_HOST_INTERFACE_EVENT_INPUT_REPORT:
        ESP_ERROR_CHECK(hid_host_device_get_raw_input_report_data(hid_device_handle,
                                                                  data,
                                                                  64,
                                                                  &data_length));

        ESP_LOG_BUFFER_HEX("Input Report", data, data_length); // Display the input report data 
        // Comment out start
        // if (HID_SUBCLASS_BOOT_INTERFACE == dev_params.sub_class) {
        //     if (HID_PROTOCOL_KEYBOARD == dev_params.proto) {
        //         hid_host_keyboard_report_callback(data, data_length);
        //     } else if (HID_PROTOCOL_MOUSE == dev_params.proto) {
        //         hid_host_mouse_report_callback(data, data_length);
        //     }
        // } else {
        //     hid_host_generic_report_callback(data, data_length);
        // }
        // Comment out stop
        break;
```

After the verification, please share the log with the result.
Thanks

Sentaku1129, 2024-08-09:
I've already tried to remove these lines of code and run them on the ESP32-S3 development board. I found that there was indeed no "STALL" error, but I still couldn't get the data from the scanner. My guess is that the scanner needs to receive the status of "NUM LOCK" and "CAPS LOCK" from the host before uploading the data, and it needs to send it to the OUT endpoint of the scanner via interrupt OUT.
I have tried to send a "set report" to endpoint 0, and the transmission error appears to be that only interrupt OUT data can be sent to endpoint OUT. What can I do? Send an interrupt OUT.

roma-jam, 2024-08-12:
Hi @Sentaku1129 , 

Usually, the Host driver notifies all the devices about NUM/CAPS/SCROLL lock state changes by using the Control Endpoint and the specific class request Set Output Report. 

Current version of the HID Host class driver doesn't provide any mechanisms to send interrupt data by OUT EP, or any other data via OUT EP. 

To have possibility to send the interrupt OUT data, the device should have such EP available. Not all devices have that and it is better to refer the manual or some kind of documentation for any particular device you would like to use, because the specific configuration can be vary. 

In any case, if the device requires some proprietary or specific logic, it is always possible to achieve the goal with using the public USB Host library API.
To check, how proprietary class driver could be designed, please refer to the `usb_host_lib` example from `esp-idf` ([here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib)).

roma-jam, 2025-01-20:
Hi @Sentaku1129 , 

is the issue still relevant? 

As it was noted before, there is no available implementation for sending the data via OUT EP in the current HID Host Class Driver. 

There are two options: to add it yourself or let us know, that the feature request is still relevant, and we will put it in our plan. 

Thanks.

roma-jam, 2025-07-09:
Hi @Sentaku1129, 

There was another issue, related to the similar problem: some devices required GetReportDescriptor() call and setting the LED status to work: https://github.com/espressif/esp-idf/issues/16005#issuecomment-2911938101

This requires adding two lines of the code: `hid_host_get_report_descriptor()` and `hid_class_request_set_report()`.  The code snippet could be found via the link I mentioned earlier.  It uses the control EP, though. 

I'll keep the issue open, because there is a possibility to add such feature, as sending the OUT report and there could be more similar requests.