espressif/esp-usb #61: Modem USB DTE: Set USB interrupt priority to ESP_INTR_FLAG_LOWMED

AguileraG, 2024-08-29:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to add LTE connectivity to an ESP32-S3 board through its USB interface. As this board makes uses of several peripherals, the CPU core on which the initialization function is executed runs out of interrupt slots with priority 1, but it still has some free slots with priorities 2 and 3.

Therefore, the USB DTE modem initialization fails, as it cannot install the USB interrupt on a slot with priority 1 ([see line 58](https://github.com/espressif/esp-usb/blob/e4817e2e9aa456e86c0037739ce087571a9242ec/host/class/cdc/esp_modem_usb_dte/esp_modem_usb.cpp#L58)). Is there any specific reason why the USB interrupt priority is set to ESP_INTR_FLAG_LEVEL1 instead of ESP_INTR_FLAG_LOWMED?

As far as I can tell from the ESP-IDF docs, all LOWMED interrupts slots should behave identically apart from their priority, and the [USB Host](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/usb_host.html) section does not mention anything related to the interrupt priority.

tore-espressif, 2024-09-02:
Hello @AguileraG , 
There is no specific reason for this priority of USB interrupt. We just tested it with this settings and it worked OK.

This section of code https://github.com/espressif/esp-usb/blob/e4817e2e9aa456e86c0037739ce087571a9242ec/host/class/cdc/esp_modem_usb_dte/esp_modem_usb.cpp#L55 was added only so you don't have to explicitly install the USB Host library, which is desired in many situations.

In your situation, you can just init `usb_config->install_usb_host = false`. This way the USB modem driver will not  install the USB host library and will not create the USB Host handling task and you can do it yourself in your application.

``` c
usb_host_config_t host_config = {};
host_config.intr_flags = your_int_priority;
usb_host_install(&host_config);
xTaskCreatePinnedToCore(usb_host_task, "usb_host", 4096, NULL, task_priority, NULL, core_id)
```

tore-espressif, 2024-09-17:
This should solve your issue. Feel free to reopen if you have any other questions

closed 2024-09-17 by tore-espressif