espressif/esp-usb #69: cdc-ncm ethernet speed hardcoded

chegewara, 2024-10-01:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
since we are having now P4 with USB-HS, is it possible to update ethernet speed notification code according to upstream repo?

https://github.com/hathach/tinyusb/discussions/742
```
tu_static struct ncm_notify_t ncm_notify_speed_change = {
    .header = {
        .bmRequestType_bit = {
            .recipient = TUSB_REQ_RCPT_INTERFACE,
            .type = TUSB_REQ_TYPE_CLASS,
            .direction = TUSB_DIR_IN},
        .bRequest = CDC_NOTIF_CONNECTION_SPEED_CHANGE,
        .wLength = 8,
    },
    .downlink = TUD_OPT_HIGH_SPEED ? 480000000 : 12000000,
    .uplink = TUD_OPT_HIGH_SPEED ? 480000000 : 12000000,
};
```
Now we are having hardcoded ethernet speed 10mb/s

Thanks

tore-espressif, 2024-10-10:
Yes, this is definitely on our todo list. Internally, we made only few P4+USB+network tests, but did not manage to make it public yet.

We will update this ticket when we have some news to share

chegewara, 2024-10-10:
Thanks
It is not that important thou.
From my test I can tell it's just cosmetic value to display in OS, it doesn't impact on real transfer speed.

peter-marcisovsky, 2025-01-09:
Looks like the ethernet speed has been updated accordingly to the `TUSB_SPEED_` in the upstream of TInyUSB already and now the changes are also present in the [`espressif/tinyusb release/v0.17`](https://github.com/espressif/tinyusb/blob/release/v0.17/src/class/net/ncm_device.c#L197-L203)

closed 2025-01-21 by chegewara