espressif/esp-usb #78: unmount function called during mount, and the mount function called during unmount

lijunru-hub, 2024-10-15:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

IDF release/v5.2

### Development Kit.

ESP32S3

### Used Component version.

1.4.5

### More Information.

The function calls are opposite to the intended logic.

lijunru-hub, 2024-10-15:
https://github.com/espressif/esp-usb/blob/master/device/esp_tinyusb/tusb_msc_storage.c#L628C1-L639C2

peter-marcisovsky, 2024-10-15:
It works like this:
`tud_umount_cb` is called when a USB device is unmounted from a USB host -> a storage is mounted on the application FW `tinyusb_msc_storage_mount`. Now the application FW owns the storage.

`tud_mount_cb` is called when a USB device is mounted to a USB host -> a storage is unmounted from the application FW `tinyusb_msc_storage_unmount`. Now the USB Host owns the storage.

https://github.com/espressif/esp-usb/blob/master/device/esp_tinyusb/include/tusb_msc_storage.h#L127

DamienEspitallier, 2024-10-21:
Is calling tinyusb_msc_storage_unmount in tud_mount_cb really correct ?

It only does something if the firmware has previously mounted the storage (and probably do something with it). It will remove the access for the firmware with no way to refuse and can happen at any moment (user can plug the cable when he wants).

May be a good way would be to add a way to refuse the unmont with a custom return value of the TINYUSB_MSC_EVENT_PREMOUNT_CHANGED callback fired in tinyusb_msc_storage_unmount ? I can do a PR if you think this is a good fix.

lijunru-hub, 2024-10-28:
Here's the translation:

There's another issue: `tusb_msc_storage.c` occupies the `tud_mount` and `tud_unmount` functions.

Although two callback interfaces are provided:
- `tusb_msc_callback_t callback_mount_changed`
- `tusb_msc_callback_t callback_premount_changed`

there is a logical bug during actual use. The callback is only triggered the first time, and subsequent plug-in or unplug events do not trigger the callback.

I believe a better approach would be to register a TinyUSB event and return the `tud_mount` and `tud_unmount` functions for user control.

peter-marcisovsky, 2025-01-17:
Hi @lijunru-hub  sorry for a late response:

To my understanding, the mount and unmount works as expected.

The following screenshot is an output log of the `esp-idf/examples/peripherals/usb/device/tusb_msc` example.

## Vbus monitor implemented (Correct example functionality)

[image]

The example works as expected. However, it is added with a [Vbus monitor](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_device.html#self-powered-device)

Now, the `tusb_msc` example looks like this  without the Vbus monitor: 

## Vbus monitor not implemented (Incorrect example functionalit)
[image]

Without the Vbus monitor, it is not possible to catch `tud_umount` callback when plugging the MSC device out of the USB host. Thus the current implementation of the `tusb_msc` example is not able to change an ownership of the storage upon the `tud_umount_cb` callback.

Does it explain the issues you are having wit the `tud_mount_cb` and `tud_umount_cb` callbacks?
We can add some notes about the Vbus monitor to the `tud_msc` example, as this is not the first time we have seen an issue connected with Vbus monitor.

@DamienEspitallier 

> May be a good way would be to add a way to refuse the unmont with a custom return value of the TINYUSB_MSC_EVENT_PREMOUNT_CHANGED callback fired in tinyusb_msc_storage_unmount ? I can do a PR if you think this is a good fix.

Thanks for pointing it out, we have been thinking about adding this functionality, so the user can actually control whether to change the storage ownership upon the USB Cable connection event to the USB Host. I will take a look at the implementation.