espressif/esp-usb #79: tinyusb_msc_storage: f_mkfs failed (14)

MatteoRodriguez, 2024-10-21:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

v5.2.1

### Development Kit.

esp32s3 custom board

### Used Component version.

v1.4.4

### More Information.

Hello,

I have a custom board with an ESP32-S3-WROOM-2 MON32R8V and an EMMC04G-MT32. I am trying to use tinyusb to make this board behave as a MSC device but I have a problem with the emmc. The tinyusb stack seems to fail to format and mount the emmc when there is no partition on it.

I am pretty sure it has nothing to do with the hardware. My whole code runs perfectly when I use the internal storage of the esp (it behaves as a MSC device). I tested the emmc example (using esp_vfs_fat_sdmmc_mount) and it can format and mount and use the emmc without any problem. I also tested the tusb_msc example and this one CANNOT format the emmc. And once the emmc has been formated, my code is working well.

[image]

As you can see on the log above, the code is able to talk with the emmc. But there is a conflict with the allocation unit size. I tried everything I could in menuconfig to set this parameter to 512 but I failed doing so. It feels to be hardcoded in the tinyusb library.

Or maybe I am missing some initializing steps with vfs_fat... Here are my initializing steps with tinyusb :

[image]

And this is the details of the storage_init_emmc function :

[image]

Thanks for your help.

tore-espressif, 2024-10-31:
@MatteoRodriguez thank you for the issue.
Indeed, we did not test USB MSC device with eMMC memory.

> And once the emmc has been formated, my code is working well.

Does this mean that you could expose the eMMC memory through USB to USB host (PC)?

igrr, 2024-10-31:
> As you can see on the log above, the code is able to talk with the emmc. But there is a conflict with the allocation unit size. I tried everything I could in menuconfig to set this parameter to 512 but I failed doing so. It feels to be hardcoded in the tinyusb library.

Could you please clarify, what leads to the conclusion that allocation unit size is a problem? Allocation unit size, also known as cluster size, is often much larger than sector size. SD cards and eMMC chips formatted by PC will often have allocation unit size (cluster size) set to 16 kB or more.

Could you please capture the logs at verbose level and upload them here as a .txt file?

MatteoRodriguez, 2024-10-31:
@tore-espressif yes.
To be more precise, if I take one of my boards with a clear emmc (no partition, nothing) and I upload either my code or the tusb_msc example code, it won't boot and keeps trying to reboot (keeps logging the content of the log file attached below).
BUT, if I upload the emmc example code, the example runs perfectly : it can boot, the emmc is fully usable and I can do everything the example is made for.
AND, if I upload again either my code or the tusb_msc example code, now it works fine because the emmc has been formatted by the emmc example code.
So there must be at least one difference between the all-in-one esp_vfs_fat_sdmmc_mount function and the way it is initialized and mounted on the tinyusb side.

@igrr well, honestly, I am pretty new to SD Card and EMMC so my only clue comes from this part of the logs :
[image]
I haven't been able to change this allocation unit parameter and test it, so it is still a lead for me, but I might be completely looking on the wrong direction. All I'm sure about is that there is a problem during either the formatting step, either during the mounting step. But I would bet on the formatting step as I don't get any errors if the emmc has already been formatted by something else.

[log.txt](https://github.com/user-attachments/files/17588699/log.txt)
Here is the log file you asked, I set every log verbosity I found in menuconfig to "verbose". Feel free to ask if you need more informations.

tore-espressif, 2024-11-01:
Hi @MatteoRodriguez,

I'm glad to hear you managed to get it working! We’ll make sure to include this in our future plans to improve the user experience for eMMC developers. However, we can’t commit to a specific timeline just yet.

MatteoRodriguez, 2024-11-04:
@tore-espressif well, I wouldn't say I got it working in a proper way as I first need to flash a code to format my emmc before I can flash my own code, but that will do the trick temporarily.

No problem, I understand about the timeline, the main goal of this issue was first to check if it wasn't due to a misunderstanding from my side, and then to let you know as soon as possible about a potential bug/feature.

I'll keep looking for a cleaner workaround with some esp_vfs_fat_sdmmc functions. I'll keep you updated. Any help/idea would be really appreciated.

igrr, 2024-11-04:
> E (3028) tinyusb_msc_storage: f_mkfs failed (14)

Error code 14 is `FR_MKFS_ABORTED,		/* (14) The f_mkfs() aborted due to any problem */` — which usually happens during argument checking in f_mkfs. Could you please revert allocation_unit_size to the default value (16kB) and try again? (I suppose you will need to wipe the contents of eMMC first.)

MatteoRodriguez, 2024-11-05:
@igrr I used the tusb_msc example and I set allocation_unit_size to 16kB but it didn't changed anything. I got the same logs as before.
[log1.txt](https://github.com/user-attachments/files/17635614/log1.txt)

metainf, 2025-02-20:
I also ran into this issue with a similar setup, when I changed `FM_FAT` with `FM_ANY`, it was able to format the eMMC. https://github.com/espressif/esp-usb/blob/aef6153faad1cd957f789a702135155ce13aae30/device/esp_tinyusb/tusb_msc_storage.c#L228

It also seems like the `mount_config` that's passed into the initial setup is not used besides setting `max_files`, and hard-coded values are used instead

erwinschrodinger1, 2025-05-12:
@tore-espressif the pr associated explains the problem and solves it with the most appropriate solution I think. If there are any better approaches to solve it can we please discuss about it? Or merge the PR?

closed 2025-05-23 by tore-espressif