espressif/esp-usb #90: Absolute Mouse movement missing

Jackolix, 2024-11-22:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The absolute mouse movement function from the tinyusb library is missing in this fork.

roma-jam, 2024-11-22:
Hi @Jackolix, 

Currently, the esp_tinyusb is using the tinyusb-based espressif component (https://github.com/espressif/tinyusb) which in it turn based on the v0.15.0 of upstream tinyusb. 

The API that were mentioned was implemented only in v0.17.0 (it is not implemented in v0.15.0: https://github.com/hathach/tinyusb/blob/0.15.0/src/class/hid/hid_device.h)

We are in the process of preparing the update and we will synchronize the tinyusb component with upstream v0.17.x. shortly.

Jackolix, 2024-11-29:
Thank you very much!