espressif/esp-usb #110: New TinyUSB teardown logic stops task but doesn't reset interrupt state

tonystuart, 2025-01-04:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

v5.3.1.1-1

### Development Kit.

Custom Board

### Used Component version.

v1.6.0

### More Information.

Based on my research, the new TinyUSB tear-down logic released in v1.6 should call also call usbd_int_set(false) to disable the interrupt that has previously been enabled.

roma-jam, 2025-01-06:
Hi @tonystuart, 

thank you for doing the research, you are right. We do need to disable interrupts and we do that.  

`usbd_int_set(false)` will lead to the `dcd_int_disable(_usbd_rhport)` which is already present in the current calling stack `tud_teardown()` -> `tud_deinit(..)` -> `dcd_int_disable(rhport)`.

So, that means, that the current implementation already disables the interrupts during the teardown, so there is no necessity to add the additional `usbd_int_set(false)` call.

roma-jam, 2025-01-20:
Hi @tonystuart , 

I will close this issue as there is nothing to add: the logic stops the task and reset the interrupt state. 

Feel free to re-open if you meant something else or create a new task if you will have further questions.

closed 2025-01-20 by roma-jam

tonystuart, 2025-01-20:
Thanks @roma-jam and apologies for not responding sooner. I had originally coded up my own tear-down logic before version 1.6 of the Espressif TinyUSB component was available. This code called tinyusb_driver_uninstall, tusb_stop_task and usbd_int_set. Later, when the tinyusb component upgraded to version 1.6 my code failed because tusb_stop_task was getting called twice. I then looked through the new version 1.6 tear-down logic and missed the macros that define _tusb_teardown_arg0/1 as tusb_rhport_teardown and filed this issue. Thanks for clarifying, it all makes sense now and is working just fine.