espressif/esp-usb #113: Number of NTB buffers configuration option

ctag-fh-kiel, 2025-01-09:
### Is your feature request related to a problem?

With the current NTB buffer settings, USB-NCM does not seem to work properly and does not achieve optimal performance.
In my setup the following error occurs quite frequently:
tud_network_can_xmit: request blocked

### Describe the solution you'd like.

Add configuration option for CFG_TUD_NCM_OUT_NTB_N and CFG_TUD_NCM_IN_NTB_N via component config.
I.e. enhancing tusb_config.h of the component with:
#define CFG_TUD_NCM_IN_NTB_N CONFIG_TINYUSB_NCM_IN_NTB_N
#define CFG_TUD_NCM_OUT_NTB_N CONFIG_TINYUSB_NCM_OUT_NTB_N

and adding to KConfig relevant lines to configure it.

peter-marcisovsky, 2025-01-09:
Hi @ctag-fh-kiel 
Thanks for the report, we will take a look. 

Could you please specify which chip and which version of Tinyusb are you using?
In the meanwhile, could you please take a look a the #107 and see if the issue is relevant to yours?

ctag-fh-kiel, 2025-01-09:
#107 may be relevant, I posted this in the relevant thread.
I am using esp32-s3 and the Tinyusb version bundled with 
  espressif/esp_tinyusb:
    version: "^1.6.0"

peter-marcisovsky, 2025-01-09:
Thank's for the info. I can observe the same behavior. 

I presume, you did try to change `CFG_TUD_NCM_IN_NTB_N` and `CFG_TUD_NCM_OUT_NTB_N` settings, If yes, how many buffers did you have to use to stop seeing the `(II) tud_network_can_xmit: request blocked` ?

In the [TInyUSB ncm.h](https://github.com/espressif/tinyusb/blob/2495563600f1cd2220d740895fad701dd48f1fb6/src/class/net/ncm.h#L63) it's stated that, for RPi RP2040, the `CFG_TUD_NCM_IN_NTB_N` shall be set to 3, to mitigate the `(II) tud_network_can_xmit: request blocked` warning

ctag-fh-kiel, 2025-01-09:
I have set
CFG_TUD_NCM_IN_NTB_N=3
CFG_TUD_NCM_OUT_NTB_N=2