espressif/esp-usb #128: is it possible to have NCM functionality along with REPL console at the the same USB port?

emaayan, 2025-01-26:
### Answers checklist.

- [x] I have read the documentation of the component in question and the issue is not addressed there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 i've been trying to do different combinations of settings to get the ability to both REPL console functionality (to be able to use putty and other serial terminals for cli) and at the same time still have USB port be recognized as a network card, but i can't seem to make it happen.

if i understand correctly one USB port can be used for serial communication (UART) which is by default used for REPL as well as uploading the firmware, the other USB port is used by default with jtag, as TUSB doesn't allow for debugging in jtag, (btw is it possible at all to debug in tinyusb mode?) it would seem to make sense that using CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG would allow me to get both console and networking on the same port. 

[image]

peter-marcisovsky, 2025-01-27:
Hi @emaayan 
On the esp32s3, there is only one USB PHY, which is shared between 2 peripherals:
- USB-OTG Peripheral
- USB Serial JTAG

Thus, you can have only one peripheral using the USB PHY at the time.
USB-OTG Peripheral can be used for running a USB application (USB Device, USB Host) or as a USB-OTG console
USB Serial JTAG Peripheral can be used for debugging or as a JTAG console. 
Docs about console [here](https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic/README.md)

In your case: You want to:
- run a USB Application (USB NCM Device)
- have a debugger
- have a repl console running

You can do that, with a following workaround:
- USB Connector on your esp32s3 board, will be occupied by the main USB Application (USB NCM Device)
- The debugger will be run from JTAG interface (Not USB Serial JTAG interface !), for that, read this [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/configure-other-jtag.html)
- The repl console environment can occupy the USB-UART Bridge (the same outuput used for flashing and monitoring) Or you can define your own custom UART Console, using a second UART peripheral (Custom UART Console) [here](https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic/main/console_example_main.c#L105)

emaayan, 2025-01-27:
> Hi [[image]@emaayan](https://github.com/emaayan) On the esp32s3, there is only one USB PHY, which is shared between 2 peripherals:
> 
> * USB-OTG Peripheral
> * USB Serial JTAG
> 
> Thus, you can have only one peripheral using the USB PHY at the time. USB-OTG Peripheral can be used for running a USB application (USB Device, USB Host) or as a USB-OTG console USB Serial JTAG Peripheral can be used for debugging or as a JTAG console. Docs about console [here](https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic/README.md)
> 
> And you can have only one functionality from one peripheral using the USB PHY at the time. Meaning: you can't have a USB application and a USB-OTG console running at the same time.
> 
> In your case: You want to:
> 
> * run a USB Application (USB NCM Device)
> * have a debugger
> * have a repl console running
> 
> You can do that, with a following workaround:
> 
> * USB Connector on your esp32s3 board, will be occupied by the main USB Application (USB NCM Device)
> * The debugger will be run from JTAG interface (Not USB Serial JTAG interface !), for that, read this [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/configure-other-jtag.html), and you will need a [ESP-PROG](https://docs.espressif.com/projects/esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html) board for debugging in this case.
> * The repl console environment can occupy the USB-UART Bridge (the same outuput used for flashing and monitoring) Or you can define your own custom UART Console, using a second UART peripheral (Custom UART Console) [here](https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic/main/console_example_main.c#L105)

actually i gave up on debugging, i'd be happy if i could get REPL console and USB-NCM in the same usb port.

peter-marcisovsky, 2025-01-27:
>  i'd be happy if i could get REPL console and USB-NCM in the same usb port

You could try creating a composite USB device with NCM and CDC interfaces and try to run some logging on the CDC interface. But I am afraid, you will not be able to run a REPL console on the TinyUSB's CDC interface.

Is there some specific reason why do you want to run repl console on the same port as your USB Device application?

emaayan, 2025-01-27:
> You could try creating a composite USB device with NCM and CDC interfaces and try to run some logging on the CDC interface. But I am afraid, you will not be able to run a REPL console on the [...]

simplicity sake, currently i'm usiing REPL to display a cli for some configuration, and the other port i'm using for network interface to view http server and stream TCP, so if people need to change the IP they need both USB, and what i am for is people be using a regular ali express dev boards with s3 .

peter-marcisovsky, 2025-01-28:
Regular esp32s3 dev boards feature both ports, if you are not targeting some of the mini series. 
Anyway, the feature you are requesting is not available right now in esp-idf and we currently don't have any plan of supporting it. If you have other questions, feel free to ask, otherwise feel free to close the issue.