espressif/esp-usb #133: task watchdog in repl console task when using tinyusb console

inbarab, 2025-02-10:
### Answers checklist.

- [x] I have read the documentation of the component in question and the issue is not addressed there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How do I run the esp-idf repl console over the tiny usb cdc-acm?
When running the esp-idf cli (esp_console_repl_task) based on the  tusb console example, I get a task watch dog on the repl task (see stack dump below)
Board: esp32-s3 wroom 1
esp-idf 5.2.3
espressif/tinyusb 0.18.0
espressif/esp_tinyusb 1.7.1

PS C:\dev\ESP\tusb_console> xtensa-esp32s3-elf-addr2line -pfiaC -e build/tusb_console.elf 0x42014FAF:0x3FC93CB0 0x420153CC:0x3FC93CD0 0x403770E9:0x3FC93D00 0x400559DD:0x3FC9CC50 0x4037D00E:0x3FC9CC60 0x4037AE7D:0x3FC9CC80 0x4200CEA5:0x3FC9CCC0 0x4200CFBB:0x3FC9CCE0 0x4200C44F:0x3FC9CD00 0x4200B55D:0x3FC9CD20 0x4200BA1E:0x3FC9CD50 0x42006B4E:0x3FC9CD70 0x4201E8EE:0x3FC9CD90 0x420098D9:0x3FC9CDB0 0x42009CE9:0x3FC9CDE0 0x4200A014:0x3FC9CE30 0x4200A051:0x3FC9CE50 0x42008D84:0x3FC9CE70 0x4037B571:0x3FC9CEA0

0x42014faf: task_wdt_timeout_handling at C:/dev/ESP/esp-idf523/components/esp_system/task_wdt/task_wdt.c:441
0x420153cc: task_wdt_isr at C:/dev/ESP/esp-idf523/components/esp_system/task_wdt/task_wdt.c:515
0x403770e9: _xt_lowint1 at C:/dev/ESP/esp-idf523/components/xtensa/xtensa_vectors.S:1240
0x400559dd: ?? ??:0
0x4037d00e: vPortClearInterruptMaskFromISR at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
(inlined by) xTaskGetSchedulerState at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/tasks.c:4982
0x4037ae7d: xQueueGenericSend at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/queue.c:941
0x4200cea5: osal_mutex_unlock at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:171
(inlined by) _ff_unlock at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/common/tusb_fifo.c:48
(inlined by) _tu_fifo_write_n at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/common/tusb_fifo.c:551
0x4200cfbb: tu_fifo_write_n at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/common/tusb_fifo.c:863
0x4200c44f: tud_cdc_n_write at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/class/cdc/cdc_device.c:178 (discriminator 4)
0x4200b55d: tud_cdc_n_write_char at C:/dev/ESP/tusb_console/managed_components/espressif__tinyusb/src/class/cdc/cdc_device.h:102
(inlined by) tinyusb_cdcacm_write_queue_char at C:/dev/ESP/tusb_console/managed_components/espressif__esp_tinyusb/tusb_cdc_acm.c:235
0x4200ba1e: tusb_write at C:/dev/ESP/tusb_console/managed_components/espressif__esp_tinyusb/vfs_tinyusb.c:127
0x42006b4e: esp_vfs_write at C:/dev/ESP/esp-idf523/components/vfs/vfs.c:482 (discriminator 4)
0x4201e8ee: write at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/syscalls/syswrite.c:11 (discriminator 1)
0x420098d9: linenoiseInsertPastedChar at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:698
0x42009ce9: linenoiseEdit at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:892
0x4200a014: linenoiseRaw at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:1105
0x4200a051: linenoise at C:/dev/ESP/esp-idf523/components/console/linenoise/linenoise.c:1168
0x42008d84: esp_console_repl_task at C:/dev/ESP/esp-idf523/components/console/esp_console_repl.c:530
0x4037b571: vPortTaskWrapper at C:/dev/ESP/esp-idf523/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

peter-marcisovsky, 2025-02-12:
Hi @inbarab 
Can you please check, if #128 is related to your case?

inbarab, 2025-02-12:
Hi @peter-marcisovsky,
I don't think it is. I'm not trying to use the same USB for several functions, I'm trying to get a single repl console to run over tinyUSB cdc-acm
This is a first step before creating a composite device (as suggested in #128) for cli + msc. That is the reasom I need the tinyusb cdc-acm, and can't use the rom usb.

I uploaded the changes I made to the example here: 
https://github.com/inbarab/esp-idf. Specifically see commit https://github.com/inbarab/esp-idf/compare/aa9b20d65de14ee3813f5cfe67a3cc8dd04e234a...cba1eebdf214ec9885faba5e25470d7b3ba56fb4

peter-marcisovsky, 2025-02-13:
Hi @inbarab 

> I'm trying to get a single repl console to run over tinyUSB cdc-acm

Please do correct me if I misunderstood your solution, but this is exactly what the other user in #128 was trying to do (the only difference being, he wanted to have a Composite USB device)

> This is a first step before creating a composite device (as suggested in https://github.com/espressif/esp-usb/issues/128) for cli + msc

Plus, your next step will be exactly the same as the other user's (creating a composite USB device)

For now, our resolution stands as in the mentioned GH Issue: 
> the feature you are requesting is not available right now in esp-idf

We can reconsider whether to officially support this feature and may put it into our development plans, since there are multiple users asking about this feature, but unfortunately it is not our priority right now and I can't say when an whether we will officially support it.

For now, you can build your solution upon [this](https://github.com/espressif/esp-usb/issues/128#issuecomment-2619079527) reply.

inbarab, 2025-02-13:
Hi @peter-marcisovsky 

Perhaps I misunderstood, but I though #128's main problem was the coexistence of the debugger and tinyusb over the usb connector
In any case, I have already used [this](https://github.com/espressif/esp-usb/issues/128#issuecomment-2619079527) as you suggest, only to run into further problems that I am addressing with espressif  support, mainly that the repl console thread blocks and triggers the task watch dog
You can reproduce this problem using the modified tusb console example from 
[here](https://github.com/inbarab/esp-idf/compare/aa9b20d65de14ee3813f5cfe67a3cc8dd04e234a...cba1eebdf214ec9885faba5e25470d7b3ba56fb4)

Thanks
Inbar

lijunru-hub, 2025-03-21:
Using `esp_console_new_repl_uart` does indeed trigger the watchdog, as the console currently does not have good support for TinyUSB CDC-ACM.  

You can:  
1. Disable the task watchdog.  
2. Use a third-party CLI. You can refer to this example:  
   [https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/device/usb_dongle/main/data_back.c](https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/device/usb_dongle/main/data_back.c).

inbarab, 2025-03-25:
I tired the example you suggested
It doesn't work either - I'll open an issue on that example :-)
In short - the function that's supposed to write the output does not because the define in that function does not exist, and once that is solved, the Command_Parse input index is not static so the command is, in fact, deleted every time a new character is received
It also doesn't echo
> https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/device/usb_dongle/main/data_back.c.

Regarding the suggestion to disable the watch dog, that does not help. While the watchdog stops tripping, the output does not work (printing from another task while waiting for input does not reach the console)
Please advise

lijunru-hub, 2025-03-28:
@peter-marcisovsky Can we make `esp_console` support TinyUSB CDC ACM?

peter-marcisovsky, 2025-03-28:
Hi @lijunru-hub  this feature is a nice to have one, but it's not among our priorities right now. Unfortunately, I can't commit to any specific date about delivering this feature.

inbarab, 2025-03-30:
I have a suspicion: is it possible the cdc-acm driver does not feed the incoming chars to the virtual file system rx fifo?