espressif/esp-usb #136: Tiny USB cannot be started after CDC-USB console is running

inbarab, 2025-02-20:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

5.2.3

### Development Kit.

ESP32-s3-wroom-1

### Used Component version.

1.7.1

### More Information.

When running the example tusb_msc, if the esp-idf console was set to USB-CDC, the tusb-msc device will get an error
E (964) intr_alloc: No free interrupt inputs for USB interrupt (flags 0xE)
I cannot find any method for disabling the USB Console and freeing the interrupts
Using the UART console is not an option in my system - I need the USB console, and the ability to replace it with the tiny USB MSC
Please advise
Inbar

peter-marcisovsky, 2025-02-20:
Hi @inbarab 
How did you change the console type from the Default UART console to USB-CDC?
There is a macro in the example, which will to allow you to set the USB-CDC console during build.

The error, you are suppose to get after changing the console type looks like this:

```
/home/peter/esp/esp-idf/examples/peripherals/usb/device/tusb_msc/main/tusb_msc_main.c: In function 'app_main':
/home/peter/esp/esp-idf/examples/peripherals/usb/device/tusb_msc/main/tusb_msc_main.c:467:2: error: #error Unsupported console type
  467 | #error Unsupported console type
      |  ^~~~~
```

inbarab, 2025-02-21:
I'm running the esample from esp-idf 5.2.3
The line "Unsupported console type" does not appear in that example
I couldn't find any #error in tusb_msc_main.c

inbarab, 2025-02-21:
I'm running the tusb_msc example from esp-idf 5.2.3 on esp32-s3
I can't find the #error you mentioned

peter-marcisovsky, 2025-02-24:
Sorry, I did not notice the the IDF version you are using. I am using master branch.

Starting in `IDF 5.3`, we introduced a [check](https://github.com/espressif/esp-idf/blob/release/v5.3/examples/peripherals/usb/device/tusb_msc/main/tusb_msc_main.c#L431-L443) for selected console type (mainly due to support for esp32p4). But it can also be used on other USB-OTG capable targets. Thus your IDF version, being 5.2.x, does not feature this check. 

Bottom line being: you can't use both, a TinyUSB device and a USB-CDC console (the one present in ROM) on esp32s3 (esp32s2). Because each of this software components uses USB PHY exclusively and both of the chips features only one internal USB PHY.

The same goes for USB Serial JTAG console on esp32s3.

inbarab, 2025-02-24:
Hi
I understand the two - ROM USB CLI and TinyUSB MSD - can't coexist, I want to **swap** between them:
Start the system with CLI
Given a cli command, shut down the ROM USB CLI and start the TinyUSB MSC
Is there a way to do that?
I can't find any documentation that tells how to release the interrupts from the ROM USB