espressif/esp-usb #140: UVC video format

JunOllyLi, 2025-02-23:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

host/uvc

### ESP-IDF version.

5.3

### Development Kit.

ESP32-S3-WROOM

### Used Component version.

2.1.0

### More Information.

I have a UVC device, from USBView on Windows, it reported supporting a lot of videos formats. But on the ESP, I was only able to get 640x480 15fps work (and the 15fps was not even in the list of supported format). 
When I chose other 

 it always complains about "uvc: uvc_claim_interface(462): Could not find frame format 480x320@10.0FPS
E (17480) uvc: uvc_host_stream_open(653): Could not find/claim streaming interface"
Any suggestions?
Attached here is the USBView output  I captured on Windows.
[USBViewAll.txt](https://github.com/user-attachments/files/18933757/USBViewAll.txt)

tore-espressif, 2025-02-24:
Hi @JunOllyLi ,
usually, USB cameras support different format on HighSpeed (that is with your WIndows PC) and different set of format for FullSpeed (that is ESP32-S3). You can only use formats that your camera supports in FullSpeed mode

JunOllyLi, 2025-02-24:
> Hi [@JunOllyLi](https://github.com/JunOllyLi) , usually, USB cameras support different format on HighSpeed (that is with your WIndows PC) and different set of format for FullSpeed (that is ESP32-S3). You can only use formats that your camera supports in FullSpeed mode

I see. It is just hard to imagine it supports 640x480@15fps instead of much lower resolution video formats with slower speed... Thanks!

tore-espressif, 2025-02-24:
Yep, sometimes it indeed is like that. Camera vendors do not care about FullSpeed operation much, sometimes the cameras do not work at all at FS, sometimes there is only 1 format available, and on some rare occasions, you get pretty nice features even at full speed