espressif/esp-usb #146: 16 bit Scroll wheel

TypingPenguin, 2025-03-09:
### Answers checklist.

- [x] I have read the documentation of the component in question and the issue is not addressed there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am having a bit of trouble transforming my HID description report to a 16 bit report (I'm new to this so it's probably me doing something wrong).
My HID looks like this (version A):
``` 
// Mouse Report Descriptor Template
#define TUD_HID_REPORT_DESC_MOUSE(...) \
HID_USAGE_PAGE ( HID_USAGE_PAGE_DESKTOP      )                   ,\
HID_USAGE      ( HID_USAGE_DESKTOP_MOUSE     )                   ,\
HID_COLLECTION ( HID_COLLECTION_APPLICATION  )                   ,\
  /* Report ID if any */\
  __VA_ARGS__ \
  HID_USAGE      ( HID_USAGE_DESKTOP_POINTER )                   ,\
  HID_COLLECTION ( HID_COLLECTION_PHYSICAL   )                   ,\
    /* Even for a scroll-only device, including X/Y with 0 values helps compatibility */\
    HID_USAGE_PAGE  ( HID_USAGE_PAGE_DESKTOP )                   ,\
      /* X, Y position - required for mouse compatibility */ \
      HID_USAGE       ( HID_USAGE_DESKTOP_X                    ) ,\
      HID_USAGE       ( HID_USAGE_DESKTOP_Y                    ) ,\
      HID_LOGICAL_MIN ( 0                                      ) ,\
      HID_LOGICAL_MAX ( 0                                      ) ,\
      HID_REPORT_COUNT( 2                                      ) ,\
      HID_REPORT_SIZE ( 8                                      ) ,\
      HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_RELATIVE ) ,\
      /* Vertical wheel [-127, 127] */ \
      HID_USAGE       ( HID_USAGE_DESKTOP_WHEEL                )  ,\
      HID_LOGICAL_MIN ( -127                                )  ,\
      HID_LOGICAL_MAX ( 127                                 )  ,\
      HID_REPORT_COUNT( 1                                      )  ,\
      HID_REPORT_SIZE ( 8                                    )  ,\
      HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_RELATIVE )  ,\
      /* Resolution Multiplier Feature (1-120) */ \
      HID_USAGE_PAGE  ( HID_USAGE_PAGE_DESKTOP                   ), \
      HID_USAGE       ( HID_USAGE_DESKTOP_RESOLUTION_MULTIPLIER ), \
      HID_LOGICAL_MIN ( 0                                      ), \
      HID_LOGICAL_MAX ( 120                                    ), \
      HID_PHYSICAL_MIN( 1                                      ), \
      HID_PHYSICAL_MAX( 120                                    ), \
      HID_REPORT_SIZE ( 8                                      ), \
      HID_REPORT_COUNT( 1                                      ), \
      HID_FEATURE     ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE ), \
  HID_COLLECTION_END                                           , \
HID_COLLECTION_END
``` 

When transforming this to get 16 bit support, I would assume I'd need to create the following (version B):
```
// Mouse Report Descriptor Template
#define TUD_HID_REPORT_DESC_MOUSE(...) \
HID_USAGE_PAGE ( HID_USAGE_PAGE_DESKTOP      )                   ,\
HID_USAGE      ( HID_USAGE_DESKTOP_MOUSE     )                   ,\
HID_COLLECTION ( HID_COLLECTION_APPLICATION  )                   ,\
  /* Report ID if any */\
  __VA_ARGS__ \
  HID_USAGE      ( HID_USAGE_DESKTOP_POINTER )                   ,\
  HID_COLLECTION ( HID_COLLECTION_PHYSICAL   )                   ,\
    /* Even for a scroll-only device, including X/Y with 0 values helps compatibility */\
    HID_USAGE_PAGE  ( HID_USAGE_PAGE_DESKTOP )                   ,\
      /* X, Y position - required for mouse compatibility */ \
      HID_USAGE       ( HID_USAGE_DESKTOP_X                    ) ,\
      HID_USAGE       ( HID_USAGE_DESKTOP_Y                    ) ,\
      HID_LOGICAL_MIN ( 0                                      ) ,\
      HID_LOGICAL_MAX ( 0                                      ) ,\
      HID_REPORT_COUNT( 2                                      ) ,\
      HID_REPORT_SIZE ( 8                                      ) ,\
      HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_RELATIVE ) ,\
      /* Vertical wheel [-32767, 32767] */ \
      HID_USAGE       ( HID_USAGE_DESKTOP_WHEEL                )  ,\
      HID_LOGICAL_MIN ( -32797                                )  ,\ <---- Changed
      HID_LOGICAL_MAX ( 32797                                 )  ,\ <---- Changed
      HID_REPORT_COUNT( 1                                      )  ,\
      HID_REPORT_SIZE ( 16                                    )  ,\ <---- Changed
      HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_RELATIVE )  ,\
      /* Resolution Multiplier Feature (1-120) */ \
      HID_USAGE_PAGE  ( HID_USAGE_PAGE_DESKTOP                   ), \
      HID_USAGE       ( HID_USAGE_DESKTOP_RESOLUTION_MULTIPLIER ), \
      HID_LOGICAL_MIN ( 0                                      ), \
      HID_LOGICAL_MAX ( 120                                    ), \
      HID_PHYSICAL_MIN( 1                                      ), \
      HID_PHYSICAL_MAX( 120                                    ), \
      HID_REPORT_SIZE ( 8                                      ), \
      HID_REPORT_COUNT( 1                                      ), \
      HID_FEATURE     ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE ), \
  HID_COLLECTION_END                                           , \
HID_COLLECTION_END
```

I used the first description with the following code (version A). This works fine with HID version A:
```
 void send_high_res_mouse_report(int16_t x, int16_t y, int16_t vertical_wheel, int16_t horizontal_wheel) {
    // 4-byte report: 1 byte X, 1 byte Y, 2 bytes vertical wheel
    uint8_t report[3] = {0};

    // X and Y positions (8-bit)
    report[0] = (uint8_t)x;
    report[1] = (uint8_t)y;
    report[2] = (uint8_t)vertical_wheel;
  
    tud_hid_report(HID_ITF_PROTOCOL_MOUSE, report, sizeof(report));    
}
```

Then I transform it to the following while also transforming the HID to version B:
```
 void send_high_res_mouse_report(int16_t x, int16_t y, int16_t vertical_wheel, int16_t horizontal_wheel) {
    
    // 4-byte report: 1 byte X, 1 byte Y, 2 bytes vertical wheel
    uint8_t report[4] = {0};  //<---- Changed

    // X and Y positions (8-bit)
    report[0] = (uint8_t)x;
    report[1] = (uint8_t)y;

    // Vertical wheel (16-bit)
    report[2] = (uint8_t)(vertical_wheel & 0xFF);  //<---- Changed
    report[3] = (uint8_t)((vertical_wheel >> 8) & 0xFF);  //<---- Changed
   
    tud_hid_report(HID_ITF_PROTOCOL_MOUSE, report, sizeof(report));    
}
```
But with version B, nothing works... And I also don't really know how to debug as I can't find the device in my device manager. This means I can't see any error that might be related to my implementation.

I am using esp-idf + tinyUSB on an esp32s3 with a AS5600 magnetic encoder.

Any help to transform it to 16 bit values would be great!

peter-marcisovsky, 2025-03-14:
Hi @TypingPenguin 

Could you try using `HID_LOGICAL_MIN_N` and `HID_LOGICAL_MAX_N` instead of `HID_LOGICAL_MIN` and `HID_LOGICAL_MAX` ?

Where in your case it would be something like:
 `HID_LOGICAL_MIN_N(0x7FFF, 2)`
 `HID_LOGICAL_MIN_N(0x8001, 2)`

peter-marcisovsky, 2025-03-31:
Hi @TypingPenguin  any update on this issue?

peter-marcisovsky, 2025-04-04:
Closing the issue due to inactivity. Please open a new one, if your problem persists.

closed 2025-04-04 by peter-marcisovsky