espressif/esp-usb #159: esp_tinyusb v1.7.3 aborts build due to default value of CONFIG_TINYUSB_MSC_BUFSIZE

atanisoft, 2025-04-15:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

5.4.1

### Development Kit.

ESP32-S3

### Used Component version.

1.7.3

### More Information.

By default `CONFIG_TINYUSB_MSC_BUFSIZE` gets assigned based on the IDF_TARGET with the ESP32-S3 set to 512b. Unfortunately this doesn't match the default flash wear leveling setting leading to compilation failure:
```
firmware/managed_components/espressif__esp_tinyusb/tusb_msc_storage.c:38:2: error: #error "CONFIG_TINYUSB_MSC_BUFSIZE must be at least at the size of CONFIG_WL_SECTOR_SIZE"
   38 | #error "CONFIG_TINYUSB_MSC_BUFSIZE must be at least at the size of CONFIG_WL_SECTOR_SIZE"
      |  ^~~~~

```

From the default generated sdkconfig:
```
#
# Wear Levelling
#
# CONFIG_WL_SECTOR_SIZE_512 is not set
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
# end of Wear Levelling
...
#
# Massive Storage Class (MSC)
#
CONFIG_TINYUSB_MSC_ENABLED=y
CONFIG_TINYUSB_MSC_BUFSIZE=512
CONFIG_TINYUSB_MSC_MOUNT_PATH="/data"
# end of Massive Storage Class (MSC)
```

The default value should be updated to match CONFIG_WL_SECTOR_SIZE instead of specifying a value based on IDF_TARGET. This validation could also be moved away from a compilation failure to a runtime validation failure when the actual APIs are called. In my code I'm not trying to mount flash / sdcard / etc but instead presenting a fully virtual file system over MSC.

tore-espressif, 2025-04-16:
@atanisoft thanks for the report, will fix ASAP

peter-marcisovsky, 2025-04-16:
The build error you are seeing, has been reported [here](https://github.com/espressif/esp-idf/issues/14566) and was fixed recently in #157 

The `CONFIG_WL_SECTOR_SIZE_512` is set by default in tusb_msc example [sdkconfig.defaults](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/device/tusb_msc/sdkconfig.defaults#L10), but in your case the WL sector is set to 4096.

Which example are you using?

atanisoft, 2025-04-16:
> Are you setting the `CONFIG_WL_SECTOR_SIZE` to 4096, while keeping the `CONFIG_TINYUSB_MSC_BUFSIZE` a the default value and not seeing the sector failed error as reported in the linked issue?

I'm using defaults for both settings as generated by ESP-IDF v5.4.1 and tinyusb / esp_tinyusb. This results in out-of-box defaults that will cause this failure.

Note that I'm not using any of the examples, etc. This is a typical ESP-IDF project that happens to enable MSC but does *NOT* use *ANY* API calls from esp_tinyusb related to MSC as I'm not after managing to expose SD/flash over MSC. I'm enabling this option *ONLY* to enable MSC features in the tinyusb config which otherwise would not be enabled.

The use case I'm using with a pure virtual disk option is to expose internal settings as a virtual file on the disk and to allow users to drag and drop new firmware onto the disk which triggers an OTA like flow using only a bare bin file and not UF2 etc.  I had originally implemented this functionality years back by wrapping TinyUSB myself in [esp32usb](https://github.com/atanisoft/esp32usb) but opted to switch to esp_tinyusb instead as it required less work to get things up and running with ESP-IDF v5.x.

> EDIT: Meanwhile, the 1.7.3 version was yanked

Thanks, I'll likely pin to an older version that disables this specific check or modify sdkconfig.defaults to override CONFIG_TINYUSB_MSC_BUFSIZE.

peter-marcisovsky, 2025-04-16:
Thanks for the explanation. 
I missed your use case. It would be better, as you are suggesting, to move away from the build time error, to run time check. Will fix the problem, sorry for the inconvenience.

peter-marcisovsky, 2025-04-17:
Hi @atanisoft could you please take a look, if the solution in #161 would work for you? Thank you.

atanisoft, 2025-04-17:
From a very quick review of the PR I think this should work. I'll pull down the diffs and apply them locally if I can later today to further confirm.

atanisoft, 2025-04-18:
I was able to confirm that this works well for the use case I'm using. Thansk!

peter-marcisovsky, 2025-04-22:
Hi @atanisoft  thank you for confirmation. We will proceed with the PR and release.

PR #161 (2025-04-25): feat(tusb_msc): Add WL Sector runtime check during spiflash init
## Description

Add Wear leveling sector size runtime check.

As the wear leveling sector size is only relevant to spiflash storage type (not SDMMC storage), the WL Sector size is checked during the `tinyusb_msc_storage_init_spiflash()`.

The `CONFIG_WL_SECTOR_SIZE` (being 4096 by default, but set to 512 using `sdkconfig.defaults` in [tusb_msc example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/device/tusb_msc/sdkconfig.defaults#L10)) must be lower or equal to the `CONFIG_TINYUSB_MSC_BUFSIZE`.

## R [... cut]