espressif/esp-usb #173: UAC 2.0 support in usb_host_uac?

eggfly, 2025-05-21:
### Is your feature request related to a problem?

As title, is it only software problem or USB 1.0 speed limit?

leeebo, 2025-05-21:
UAC 2.0 has made significant improvements in the descriptor format and the transmission mode of data packets. We will consider supporting it

confusedsakata, 2025-11-17:
I also need 2.0 support so I would greatly appreciate it if it was added.

> I'd like this as well. I am very likely going to be working on this myself since there doesn't seem to be a solution available. If I do so, can I just submit a pull request?

Please do! Would appreciate it.

Averyy, 2026-10-29:
I wrote a UAC2 host driver for the ESP32-S3 that might help here: https://github.com/Averyy/esp-uac2-host. It's MIT, it's on the registry as `averyy/usb_host_uac2`, and it's modeled on `usb_host_uac`, so the API should feel familiar. It handles clock selectors, async feedback, 44.1 kHz and hot-unplug.

On the original question: UAC2 works fine at Full Speed. It's just missing software.

Most of my testing is with a miniDSP 2x4 HD on IDF 5.4 through 6.1. I haven't tried a P4.

If you'd like it in esp-usb, I'm happy to put together a PR (USB-173), as its own component or folded into `usb_host_uac`, whichever you prefer.

self-destruction, 2026-11-02:
Since UAC2 support now exists as a third-party component, my question is about the host stack itself: does the current esp-usb USBH support isochronous OUT transfers together with an async feedback endpoint at High Speed on ESP32-P4 (125 µs interval)?

Is there any example of full-duplex audio (ISO IN + ISO OUT) at HS?
I'm going to port a UAC2 host driver to the P4 and want to know what to expect from the transport layer.