espressif/esp-usb #179: Splitting each component in this repo into separate repo and use submodules

second-string, 2025-05-24:
### Is your feature request related to a problem?

The package manager for the `esp_tinyusb` component ([here](https://components.espressif.com/components/espressif/esp_tinyusb)) links directly to the `esp-usb/device/esp_tinyusb` directory. This repo also has multiple other components located in the `esp-usb/host` directory.

If I add `esp_tinyusb` as a component in my project, the component manager downloads and uses just the `esp-usb/device/esp_tinyusb` directory. For a couple reasons I would like to fork the `esp_tinusb` component and use it as a git submodule in my own components directory. However right now, that requires me to fork the entire `esp-usb` repo and add that as a submodule in my components directory. That leaves me with a lot of code I don't want, and a directory structure I don't need. I think I'd also have to manually add the recursive component paths, since I'm not sure the regular esp-idf build process looks past the first level when searching for components in the components directory.

### Describe the solution you'd like.

It would be nice if all of the components in this repo lived in their own repos as a top-level component. This would allow users to pick and choose the components they would like to fork and edit, then include directly as submodules.

This main `esp-usb` repo could still be a collection of all of the components, but they would just be submodules in this repo instead of directly included.

I recognise this is more effort for maintaining and upstreaming changes, but I thought the whole point of the new component system was to separate out components to allow quick updating, instead of have to bulk update the main repo (`idf`, or in this case `esp-usb`) once changes to sub-components are made.

If I'm missing an easier way to access just the `esp_tinyusb` component _without the component manager_, then please let me know!

peter-marcisovsky, 2025-05-27:
Hi @second-string  thanks for the request. We will consider your suggestions and let you know.

second-string, 2025-06-12:
Understood, thanks for the workaround.