espressif/esp-usb #221: some DRP examples with for example WUSB3801 for cc detection and role detection

savenlid, 2025-08-07:
### Describe the solution you'd like.

Some examples where the wusb3801 ( or similar ) will detect the usb-c situation with help of cc-lines

and do full drp role.

if device is detected, turn on vbus.
and identify the device.

if a host is detected, instantiate the usb-ncm function.

preferably for the esp-p4.

I am myself on may way to do this.

Hardware with P4+WUSB3801 can be provided if you want.

tore-espressif, 2025-08-11:
Hello @savenlid ,

providing better support for USB Type-C designs is currently one of our top priorities.
Internally, we work on development board based on FUSB302 and HUSB320 chips. We plan to add support for WUSB3801, TUSB320 and similar chips in the future.

We will keep this ticket updated once we have some new information

savenlid, 2025-08-27:
I see now the msc example is updated for 5 months ago.

Updates seems somewhat related to my problem.
I will see if it helps.

savenlid, 2025-09-13:
@roma-jam 

FYI on the unrelated issue, I got it working.

you need to tear down dhcp-s explicitly.
    err = esp_netif_dhcps_stop(esp_netif_p);
    esp_netif_destroy(esp_netif_p); 

destroy of the complete if still leaves dhcp-data-structures etc intact in memory.
possible even allocated.
at the next connect it got reused, and it was already "connected" , it saw no need to give a dhcp-adress again.

my personal opinion is that bring down the if should bring down associated stuff also.

savenlid, 2026-10-15:
if you want to you can have some example-code

i run this since a long time now, DRP, and it works.