espressif/esp-usb #236: TinyUSB component package version doesn't match documentation

projectgus, 2025-08-28:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

v5.4.2

### Development Kit.

ESP32-S3-DevKitC

### Used Component version.

1.7.6~1 with tinyusb component 0.18.0~4

### More Information.

This issue report relates to the `tinyusb` component, not `esp_tinyusb`, but that repo has GitHub issues disabled.

The documentation for [ESP-IDF Component Manager](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/versioning.html) says that the "Revision" version increments when "there are only downstream code changes between two revisions." This is not the case for recent revision increments of the TinyUSB component which contain unreleased breaking changes from upstream TinyUSB.

For example, TinyUSB commit https://github.com/hathach/tinyusb/commit/db537861b2e39debe09009762a968edfcd4a9618#diff-d34d3b84d6fa19c64b96ed57ceae385cfddea1c7e74e76f55ffe61f2eb2ddf2cR54-R67 makes a backwards incompatible behaviour change (new flag `tx_overwritabe_if_not_connected`). This change is not yet released in any TinyUSB version, but it's included in Espressif downstream releases `v0.18.0~3` and `v0.18.0~4`. According to the versioning documentation this shouldn't happen.

This new flag has caused a regression in MicroPython (the CDC FIFO mode becomes blocking). It's hard for us to work around it because according to the [TinyUSB version macros](https://github.com/espressif/tinyusb/blob/1a783b357343d053eb861ae783922371ce9e1b17/src/tusb_option.h#L33) this is TinyUSB v0.18.0. This is clearly not true, this version is *actually* a random snapshot of TinyUSB upstream master branch.

If it's necessary to release components based on unreleased upstream TinyUSB then suggest tagging them as `v0.19.0-dev0` or similar (with a change in the TUSB_VERSION macros as well, if possible, for projects like MicroPython that also use upstream TinyUSB). This would be compatible with ESP-IDF's own component version documentation, and make it easier for users to understand that there may be new - possibly unstable - code changes included from upstream.

tore-espressif, 2025-09-03:
Hello Angus,

thank you for pointing out the problem, and sorry for the trouble it has caused.
We intended to provide ESP users with more frequent TinyUSB updates, but we overlooked this breaking change. Since upstream TinyUSB does not release often enough to keep pace with our new chip support, our plan is to start using our own versioning scheme. This should help us avoid similar issues in the future.

That said, we would like to avoid yanking the already published version 0.18.0~4.
Would it be acceptable for you to pin TinyUSB to a specific version in MicroPython?
Going forward, we will release 0.19 and later versions in sync with upstream TinyUSB releases.

_(Note: v0.19.0-dev0 is a pre-release version, so version solvers will not select it by default.)_

projectgus, 2025-09-09:
@tore-espressif Thanks for the quick and thoughtful response, I appreciate your explanation. I didn't expect that you'd yank the version or anything like that, I only wanted to make you aware for future releases.

We will indeed pin the version in MicroPython now (and this was a good reminder to be vigilant about tracking dependency lock files!)

Setting up your own Espressif TinyUSB versioning scheme seems like a good path forward.