espressif/esp-usb #256: esp_tinyusb 2.0 does not work with ESP32-P4 USB1.1 peripheral

tore-espressif, 2025-09-15:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/esp-usb/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

device/esp_tinyusb

### ESP-IDF version.

master

### Development Kit.

P4 test board

### Used Component version.

2.0.0

### More Information.

Application fails here https://github.com/espressif/esp-usb/blob/master/device/esp_tinyusb/descriptors_control.c#L94

In previous 1.x versions, `tud_descriptor_other_speed_configuration_cb` was compiled only for HS configuration.
Now it always compiles for P4, even when only the FS configuration is used.

tore-espressif, 2025-09-16:
@roma-jam Quick fix is

``` c
if (!s_desc_cfg.other_speed) return NULL;
```
I can't think of any better solution that would work with our runtime peripheral selection.... :(

roma-jam, 2025-09-16:
@tore-espressif , 
yep, thanks. 

[image]

I am running the usbcv tests now, but they are not issue the get other speed if device is only FS. 

Thinking of covering it in another way.

tore-espressif, 2025-09-16:
I did some more tests, and I think it is Windows specific issue.

Because Windows always tries to fetch Device Qualifier descriptor. And if the device ACKs, it will also request Other Speed descriptor (even if the device is FS only). See this comment https://github.com/hathach/tinyusb/blob/master/src/device/usbd.c#L1162

AFAIK, linux does not fetch Device Qualifier descriptor, right?

roma-jam, 2025-09-16:
@tore-espressif, 

> AFAIK, linux does not fetch Device Qualifier descriptor, right?

Hard to say. Maybe it does depend on the build.

In my current case it does try to fetch the Device Qualifier descriptor three times for FS device (but we stall the request, because of `return (uint8_t const *)s_desc_cfg.qualifier;`). 

But there are no such requests for HS device, which is strange at first, but maybe I need to initiate it manually by requesting information of the device. At least this is not the part of "default" enumeration process. 

[image]